atomic14/esp32_audio #1: No audio Found

mlamaa, 2020-09-11:
0after running the code, and converting the raw file to wave file, there is sound at all, knowing that some data are reaching the server and the raw file is being created.

im using I2S using the same code in the last commit
```
// Direct i2s input from INMP441 or the SPH0645
  i2sSampler = new I2SMEMSSampler(i2sPins, false);

  // set up the i2s sample writer task
  TaskHandle_t i2sMemsWriterTaskHandle;
  xTaskCreatePinnedToCore(i2sMemsWriterTask, "I2S Writer Task", 4096, i2sSampler, 1, &i2sMemsWriterTaskHandle, 1);

  // start sampling from i2s device
  i2sSampler->start(I2S_NUM_0, i2sMemsConfigBothChannels, 32768, i2sMemsWriterTaskHandle);
}
```
The component connectivity ca be found  [here](https://drive.google.com/file/d/1Ihx7vVPxc_jjC3Ilvl42wLRFPKQptiRh/view?usp=sharing)

mlamaa, 2020-09-12:
I managed to play with the configuration
i changed the code as the below

> i2s_pin_config_t i2sPins = {
>     .bck_io_num = GPIO_NUM_32,
>     .ws_io_num = GPIO_NUM_25,
>     .data_out_num = I2S_PIN_NO_CHANGE,
>     .data_in_num = GPIO_NUM_35};

instead of 

> // i2s pins
> i2s_pin_config_t i2sPins = {
>     .bck_io_num = GPIO_NUM_32,
>     .ws_io_num = GPIO_NUM_25,
>     .data_out_num = I2S_PIN_NO_CHANGE,
>     .data_in_num = GPIO_NUM_33};

Now the sound contains some noise. anything than can help?

cgreening, 2020-09-12:
I think you are using the MAX9814 board - for that one you need to use the adcSampler, not the i2sSampler.

The i2sSampler will only work with microphone boards that have a built-in ADC converter and i2s interface.

The MAX9814 outputs an analogue signal so still needs to use the ESP32 analogue to digital converter. We then read samples from the ESP32 ADC using the built-in I2S peripheral so we can take advantage of DMA to transfer the data.

Use this code from `main.cpp`

```c++
  // internal analog to digital converter sampling using i2s
  // create our samplers
  adcSampler = new ADCSampler(ADC_UNIT_1, ADC1_CHANNEL_7);
  // set up the adc sample writer task
  TaskHandle_t adcWriterTaskHandle;
  xTaskCreatePinnedToCore(adcWriterTask, "ADC Writer Task", 4096, adcSampler, 1, &adcWriterTaskHandle, 1);
  adcSampler->start(I2S_NUM_0, adcI2SConfig, 32768, adcWriterTaskHandle);
```

And make sure the direct i2s input is comment out:

```c++
  // Direct i2s input from INMP441 or the SPH0645
  // i2sSampler = new I2SMEMSSampler(i2sPins, false);

  // set up the i2s sample writer task
  // TaskHandle_t i2sMemsWriterTaskHandle;
  // xTaskCreatePinnedToCore(i2sMemsWriterTask, "I2S Writer Task", 4096, i2sSampler, 1, &i2sMemsWriterTaskHandle, 1);

  // start sampling from i2s device
  // i2sSampler->start(I2S_NUM_0, i2sMemsConfigBothChannels, 32768, i2sMemsWriterTaskHandle);
```

`ADC1_CHANNEL_7` uses GPIO_NUM_35 so make sure you are not using that o

mlamaa, 2020-09-12:
It works perfectly, thank you for your clarification

closed 2020-09-12 by mlamaa

cgreening, 2020-09-12:
No problem! Good luck with your project :)

mlamaa, 2020-09-12:
How can i make the recording the specific period of time, lets same 10 second?

scottky00, 2022-10-22:
Hi, has anything in the code changed the setup for this? adcSampler->start no longer takes those arguments and I am trying to get code for serializing these packets of audio data working on an ESP32.

My current initialization looks like this
// printf("adcSampler intialize\n");
  adcSampler = new ADCSampler(ADC_UNIT_1, ADC1_CHANNEL_7, adcI2SConfig);
  // printf("adcSampler created\n");
  // set up the adc sample writer task
  TaskHandle_t adcWriterTaskHandle;
  // printf("Begin pinned to core task\n");
  xTaskCreatePinnedToCore(adcWriterTask, "ADC Writer Task", 4096, adcSampler, 1, &adcWriterTaskHandle, 1);
  adcSampler->start();

The configuration of my pins is
i2s_pin_config_t i2sPins = { 
    .bck_io_num = GPIO_NUM_32,
    .ws_io_num = GPIO_NUM_25,
    .data_out_num = I2S_PIN_NO_CHANGE,
    .data_in_num = GPIO_NUM_35};

And my i2s config is
i2s_config_t adcI2SConfig = {
    .mode = (i2s_mode_t)(I2S_MODE_MASTER | I2S_MODE_RX | I2S_MODE_ADC_BUILT_IN),
    .sample_rate = 16000,
    .bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT,
    .channel_format = I2S_CHANNEL_FMT_ONLY_LEFT,
    .communication_format = I2S_COMM_FORMAT_I2S_LSB,
    .intr_alloc_flags = ESP_INTR_FLAG_LEVEL1,
    .dma_buf_count = 4,
    .dma_buf_len = 1024,
    .use_apll = false,
    .tx_desc_auto_clear = false,
    .fixed_mclk = 0};

I get data out of the serial port but it looks like garbage/bad data, I was wondering if maybe I was missing something simple in my setup, I can send more code if needed. Thank you!