atomic14/esp32_audio #2: Noise adjustment.

john-dvp0, 2020-09-13:
Hello! Your project helped me a lot. But I need to adjust the noise. The recording comes out paused and with a little noise.

My board:
ESP32 ---- MAX9814
3v3................Vdd
GND.............GND
GPIO35........Out
GND..............Gain - 40dB

Image of the audio output.
[image]

Video in zip.
[ScreenRecorderProjectAudio.zip](https://github.com/atomic14/esp32_audio/files/5215070/ScreenRecorderProjectAudio.zip)

cgreening, 2020-09-13:
There's probably a couple of issues.

The main issue I think you are seeing is the same problem I had with noise coming through from the power supply when the ESP32 is sending data over WiFi.

The other issues may be with trying to send data too quickly - sampling at 40KHz means the ESP needs to send a lot of data very quickly to the server and it may overload.

Try decreasing the sample rate to 20KHz and see what effect that has.

john-dvp0, 2020-09-14:
Hello! I did the test at 20KHz. The audio comes out fast. Video in the zip file.

[ScreenRecorderProjectAudio20KHz.zip](https://github.com/atomic14/esp32_audio/files/5218955/ScreenRecorderProjectAudio20KHz.zip)

cgreening, 2020-09-14:
What are you using to import the raw data?

You'll need to make sure you tell it the sample rate and data format.

john-dvp0, 2020-09-14:
Perfect! I used AudaCity to make the adjustments. Now only the noise remains.
[ScreenRecorderProject20KHz_V2.zip](https://github.com/atomic14/esp32_audio/files/5219222/ScreenRecorderProject20KHz_V2.zip)

cgreening, 2020-09-14:
Wow, that is pretty bad noise!

This is similar to the problem I was having in the video with noise from the power supply, but yours seems a lot worse.

These microphones do seem to be very susceptible to noise on the power line.

You can try the trick I used to get a cleaner power supply to the microphone board (assuming it's the same problem). I used an RC filter from the 5 volt line followed by a low dropout regulator to get 3v3 supply to the microphone board.

You need to use a low value for the resistor so that you still get enough current to the microphone board.

If that is not an option for you then I found the INMP441 I2S microphone to be a lot better - https://youtu.be/3g7l5bm7fZ8

john-dvp0, 2020-09-15:
Hello! I'm from Brazil, and unfortunately I don't have the INMP441 here ... I'm doing tests with a voltage regulator ... It improved a little, but there is still a lot of noise.
Voltage regulator:
[image]

cgreening, 2020-09-15:
Hi John,

Which version of the code are you using? I just noticed a bug in the latest commit that broken ADC sampling, but listening to your audio I don't think that is the problem you are seeing.

Here's is some audio I've recorded with the latest code:

https://youtu.be/isJ24Ic5sEY

As you can tell it's still quite noisy - I am running with maximum gain though.

I'm using a lot of filtering on the power supply - https://easyeda.com/editor#id=68ac790a52e544ddbfc120da932fa4e9|330d7441629d4c8598927b9622ec3343

Ignore the circuit board and just look at the schematic.

You may get some improvements by adding an LC filter on the output of your regulator.

Hope this helps a bit.
Chris

cgreening, 2020-09-21:
You may want to checkout this issues as well. Similar problems.

https://github.com/atomic14/esp32_audio/issues/3

john-dvp0, 2020-09-21:
Hello! Thank you very much .... I will try these tips. Now, I needed another help, related to sending the recording to an external server. Local testing, it's ok ... but I tried to connect to an external server and was not successful. I have a windows instance on AWS, I would like to send the recording to this server. How can I do it? If so, I can open another question for this matter. I appreciate if you can help me. Thank you!

cgreening, 2020-10-23:
Hi John,

Did you solve this issue? Provided the server is accessible over the internet then you should be able to send data to it. Just change the URL in the firmware code.

Thanks
Chris

closed 2020-10-23 by cgreening