atomic14/esp32_audio #3: A2DP source using adc dma (i2s) not working

rujha, 2020-09-17:
Hi Chris,

Thanks for your wonderful video and nice explanation related to MAX9814 and ESP-32. I am trying to send digital audio from Max9814 to bluetooth headphone ( as sink).
Source esp-32 is able to connect to sink esp-32. I am able to run the basic example over classic bluetooth and send noise audio.
But when I connect data out from MAX9814( which I have confirmed to be working) on ADC channel 0 of source esp-32 after adding I2S code, I am not getting any audio on speaker connected to sink device.

Is it possible for you provide some pointers, if any.

thanks,
Rupak

cgreening, 2020-09-17:
I've added a sample for I2S output to the repo which works with the Adafruit MAX98357A board.

It sounds like the issue might be in the communication from your source esp to the sink esp - some things to check - what channel is the sink esp using to get the data from? Maybe you are only sending data on the left channel and it is expecting data on the right channel (or vice versa).

What format is the destination expecting? You might be sending 16bits and it wants 24 or 32 bits.

That's all I can think of for now - do you have example code you can share?

rujha, 2020-09-17:
thanks for the input Chris!! able to get the data on sink esp-32. 
but it has some noise. Looks like MAX9814 is picking the noise due to power supply and sending it across.  any pointers here.

cgreening, 2020-09-17:
Sounds like you are having the same problems I had in the video.

A quick and dirty solution is to just pass the power supply through an RC filter before feeding it into the MAX9814. It doesn't draw much current ~20mA. So you can use a fairly large resistor and big capacitor.

Ideally, I would use the 5V and filter that and follow that with an LDO so you are isolated from the ESP32 noise a bit more. But if you don't have an LDO handy filtering the 3v3 output from the dev board might be good enough.

In the followup video (https://youtu.be/3g7l5bm7fZ8) I went slightly over the top and put in an LC filter, capacitor multiplier, LDO and another LC filter... but honestly, the simple RC filter seemed to do just as good a job.

cgreening, 2020-09-17:
If you can get hold of an INMP441 then it's really good at rejecting noise and is a lot quieter than the MAX9814.

rujha, 2020-09-17:
Hi Chris,
Thanks for the video with great explanation.
So as per your suggestion, should I use 3v3 output from esp-32  with RC filter to power MAX9814 to reduce the noise, as i dont have an LDO handy.

currently I don't have INMP441, will be trying that out over the weekend . Is it advisable to have RC filter to the power source for INMP441 also?

thanks,
Rupak

cgreening, 2020-09-17:
Hi Rupak,

Yes, if you don't have an LDO then you'll need to use the 3v3 output - it should work with an RC filter - try with fairly low values, to begin with - 22ohm, 33ohm, but use the largest capacitor you can.

If you have an inductor then you can add that as well in series with the resistor.

```
                   +------------+   +------------+
3v3  +-------------+inductor    +---+ resistor   +--------------------+ To mic VDD
                   +------------+   +------------+     |
                                                       |
                                                       |
                                                       |
                                                       |
                                                  +----+------+
                                                                capacitor
                                                  +----+------+
                                                       |
                                                       |
                                                       |
 GND+-------------------------------------------------------------------+

```

I found the INMP441 pretty much immune to the power noise.

rujha, 2020-09-21:
Hi Chris,

I tried with your suggestion ( LDO with resistor+ capacitor to VDD) and it reduced the noise significantly.
Still there is little noise ( something lik hissing sound) present.

5v--- 44 Ohm resistor -----LDO ------1000 Microfarad capacitor ---to Mic VDD

Is there any possibility to reduce this noise further?

On INMP441 front, I ordered it and waiting for the delivery. Once I get that, will update you on this.

thanks,
Rupak

cgreening, 2020-09-21:
That sounds like what I have as well. A very audible hiss along with the audio.

The INMP441 will solve this issue for you as it's is much less noisy.

You could try filtering the signal in the esp by averaging the samples. Or even put a low pass filter on the output of the microphone e.

But I think you may be hitting the same limits that I had with analogue microphone boards.

rujha, 2020-09-28:
Hi Chris,

I tested the same with INMP441 & SPH0645 I2S MEMS Microphone and there is no noise at all.
Thanks a lot for your suggestions.

-Rupak

rujha, 2020-10-19:
Hi Chris,

3 weeks back I tested ESP32 with INMP441 microphone . I could easily connect ESP and get perfect sound data by defining I2S communication. I have no idea what happened, but now, anything I do, I get a massive noise on top of my sound. I changed Esp32, connecting wires, INMP441 chip ( tested with 3 chips, same result) but still having the same noise issue.
Do you face this issue anytime or any pointer for the same.

Thanks,
Rupak

cgreening, 2020-10-23:
I've not experienced this with I2S microphones. Are you doing any processing of the audio on the ESP32 side of things?

Are you capturing both the LEFT and RIGHT channels and is the noise on both channels?

rujha, 2020-10-23:
After debugging I found that whenever I use (I2S_COMM_FORMAT_I2S | I2S_COMM_FORMAT_I2S_MSB)  noise is there, but If i use I2S_COMM_FORMAT_I2S then there is no noise on I2S microphone.

cgreening, 2020-10-23:
That's very strange. I will do some testing here to see if I can recreate and update the code if needed.

closed 2020-10-23 by cgreening