atomic14/esp32_audio #5: Advice on HTTP errno: 104 when using i2s_sampling and server

JRVeale, 2020-10-21:
I've been playing around with your ICS43434 breakout board (it's pretty nice to get my builtin LED flashing along with volume as a test!) and am running into some HTTP issues when trying to use your i2s_sampling and server example. I hope you might have some suggestions as this server stuff is all brand new to me.

The error I'm seeing (as reported by my ESP32) is the following on every call of `httpClient.post()` within `sendData()` (the fd cycles from 54 through to 63, and then loops).

```[E][WiFiClient.cpp:365] write(): fail on fd 54, errno: 104, "Connection reset by peer"``` 

The Express server (no changes made there - I know _nothing_ about TS / JS) gives me the below as it receives each POST request.

```
[2020-10-21T22:23:25.701Z]  "POST /i2s_samples" "ESP32HTTPClient"
[2020-10-21T22:23:28.877Z]  "POST /i2s_samples" Error (404): "Not found"
```

I'm not sure how the 104 and 404 relate, but again this is new to me. Is the issue that I'm somehow not sending `bytes` to the server?

I've made only minimal changes to your code (pinout for my board, some minor i2s_config stuff that I've verified works in other examples, fixing the thing seen in #4, etc)

Any advice greatly appreciated!

cgreening, 2020-10-22:
That's very odd. The logs you are seeing from the server don't match what I see from mine:

```
server started at http://0.0.0.0:5003
Got 32768 I2S bytes
Got 32768 I2S bytes
```

If I *don't* run the server then I see the same messages as you:

```
[E][WiFiClient.cpp:258] connect(): socket error on fd 55, errno: 104, "Connection reset by peer"
[E][WiFiClient.cpp:258] connect(): socket error on fd 56, errno: 104, "Connection reset by peer"
```

So it's a bit strange. How are you starting the server up?

JRVeale, 2020-10-22:
Interestingly, if I don't run the server I don't get any error messages from the WiFiClient (I don't get any at all!) It just hangs for a while and then tries sending data again.

I start the server up by navigating to the src folder in server in the cmd line (which has the TypeScript file and the generated Javascript file) and running "http-server". It then tells me it's available on two different IPs, in port 8080 in both (I thought it would be 5003 when reading index.ts, but I see the comment says _default_ port, so I'm not too worried there).

I start the server in this way because my googling of "how to run a node server" led me to it... Is there a more correct way?

cgreening, 2020-10-23:
Ah, sorry, that's my fault, I thought I'd added in instructions on how to start the server - I'll update the docs to make it more clear.

You just need to run the following commands:

To get it all setup run

```
cd server
yarn
```

And from then on you can just run:

```
cd server
yarn start
```

I'll add a readme to the server project and a getting started section to the main readme.

JRVeale, 2020-10-23:
Brilliant, this fixed everything. Thank you!

closed 2020-10-23 by JRVeale