atomic14/esp32_audio #6: TypeError: Cannot read property 'length' of undefined     at /Users/mac/Documents/Arduino/esp32_audio-master/server/src/index.js:22:35

mehrezboulares, 2020-10-23:


mehrezboulares, 2020-10-23:
could you please help me on it?

TypeError: Cannot read property 'length' of undefined
    at /Users/mac/Documents/Arduino/esp32_audio-master/server/src/index.js:22:35

cgreening, 2020-10-23:
This looks like you are posting something to the server that it doesn't understand. Have you modified the firmware code?

The only line in the code that uses length is:

```
console.log(`Got ${req.body.length} I2S bytes`);
```

Which should be logging out the length of the post body that was sent to the server.

mehrezboulares, 2020-10-23:
thank you for the response cgreening,

but the problem is from req.body is undefined i dont understand why?

// route to handle samples from the I2S microphones - 32 bit stereo channel samples
app.post('/i2s_samples', function (req, res) {
    // tslint:disable-next-line:no-console
    console.log("req.body "+req.body)
    console.log("Got " + req.body.length + " I2S bytes");
    fs.appendFile('i2s.wav', req.body, function () {
        res.send('OK');
    });
});

mehrezboulares, 2020-10-23:
"use strict";
exports.__esModule = true;
var express = require("express");
var body_parser_1 = require("body-parser");
var fs = require("fs");
var app = express();
var port = 10000; // default port to listen

app.use(body_parser_1["default"].raw({
    // inflate: true,
    type: '*/*'
}));

// route to handle samples from the ADC - 16 bit single channel samples
app.post('/adc_samples', function (req, res) {
    // tslint:disable-next-line:no-console
    console.log("Got " + req.body.length + " ADC bytes");
    fs.appendFile('adc.raw', req.body, function () {
        res.send('OK');
    });
});
// route to handle samples from the I2S microphones - 32 bit stereo channel samples
app.post('/i2s_samples', function (req, res) {
    // tslint:disable-next-line:no-console
    console.log("Got " + req.body.length + " I2S bytes");
    fs.appendFile('i2s.raw', req.body, function () {
        res.send('OK');
    });
});
// start the Express server
app.listen(port, '192.168.1.6', function () {
    // tslint:disable-next-line:no-console
    console.log("server started at http://192.168.1.6:" + port);
});

this is the error...

TypeError: Cannot read property 'raw' of undefined
    at Object.<anonymous> (/Users/mac/Documents/Arduino/esp32_audio-master/server/src/index.js:9:34)
    at Module._compile (internal/modules/cjs/loader.js:776:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:787:10)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
    at Function.Module._load (internal/modules/cjs/loader.js:585:3)
    at Function.Module.runMain (internal/modules/cjs/loader.js:829:12)
    at startup (internal/bootstrap/node.js:283:19)
    at bootstrapNodeJSCore (internal/bootstrap/node.js:622:3)

cgreening, 2020-10-23:
It's a little confusing as the code you've posted doesn't match up to what I see when I look in the `dist` folder at the compiled typescript. This may just be different versions of node causing the problem.

Can you walk me through the steps you are running to get the code and run it?

When you checkout the repo do you run:

```
cd server
yarn 
```

This should install all the dependencies.

And then run:

```
yarn start
```

To start the server?

mehrezboulares, 2020-10-23:
i used tsc index.ts then node index.js

mehrezboulares, 2020-10-23:
i'm beginner with nodejs... 
there is a possibility to use python flask like this?
@app.route('/i2s_samples', methods=['POST'])
def i2s_samples():
    logger.info("welcome to upload`")
    if request.method == "POST":
        f = open('i2s_samples.wav', 'wb+')
        #print(request.data)
        #f.write(request.data)
        scipy.io.wavfile.write('i2s_samples.wav', 16000, request.data)
        f.close()
    return ''

i tested it but the saved file look wrong can you help me to fix this code?

cgreening, 2020-10-23:
I think if you run with the instructions I gave you should get something that works.

```
cd server
yarn install
yarn start
```

For your python code, I think that request.data will be bytes but the data coming in will be 16-bit data. So you'll need to convert it somehow if you want to use the wavfile function.

Saving the raw data *should* work, but I'm not really an expert with flask.

mehrezboulares, 2020-10-23:
error TS18003: No inputs were found in config file '/Users/mac/Documents/Arduino/esp32_audio-master/server_/tsconfig.json'. Specified 'include' paths were '["src/**/*"]' and 'exclude' paths were '["dist"]'.

Found 1 error.
i used yarn install and yarn start bu this is the error:

error Command failed with exit code 2.
info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.
error Command failed with exit code 2.
info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.

mehrezboulares, 2020-10-23:
(esp32) MacBook-Pro-de-mac:server mac$ yarn install
yarn install v1.22.10
[1/4] 🔍  Resolving packages...
success Already up-to-date.
✨  Done in 0.13s.
(esp32) MacBook-Pro-de-mac:server mac$ yarn start
yarn run v1.22.10
$ yarn run build
$ tslint -c tslint.json -p tsconfig.json --fix
$ tsc
src/index.ts:5:13 - error TS2349: This expression is not callable.
  Type 'typeof e' has no call signatures.

5 const app = express();
              ~~~~~~~

  src/index.ts:1:1
    1 import * as express from 'express';
      ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    Type originates at this import. A namespace-style import cannot be called or constructed, and will cause a failure at runtime. Consider using a default import or import require here instead.

src/index.ts:16:27 - error TS7006: Parameter 'req' implicitly has an 'any' type.

16 app.post('/adc_samples', (req, res) => {
                             ~~~

src/index.ts:16:32 - error TS7006: Parameter 'res' implicitly has an 'any' type.

16 app.post('/adc_samples', (req, res) => {
                                  ~~~

src/index.ts:25:27 - error TS7006: Parameter 'req' implicitly has an 'any' type.

25 app.post('/i2s_samples', (req, res) => {
                             ~~~

src/index.ts:25:32 - error TS7006: Parameter 'res' implicitly has an 'any' type.

25 app.post('/i2s_samples', (req, res) => {
                                  ~~~

Found 5 errors.

error Command failed with exit code 2.
info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.
error Command failed with exit code 2.
info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.

cgreening, 2020-10-23:
I've just pushed up a new version of `package.json` to make sure it uses the locally installed version of tsc.

Can you pull it down and see if it works.

This is what you should be seeing:

```
/tmp/esp32_audio/server (master)$ yarn install
yarn install v1.22.5
[1/4] 🔍  Resolving packages...
success Already up-to-date.
✨  Done in 0.12s.
/tmp/esp32_audio/server (master)$ yarn start
yarn run v1.22.5
$ yarn run build
$ tslint -c tslint.json -p tsconfig.json --fix
$ tsc
$ node .
server started at http://0.0.0.0:5003
```

mehrezboulares, 2020-10-25:
many thanks cgreening, now the server is working...
but can you please give me an ida how i can read the generated raw files ???

mehrezboulares, 2020-10-25:
the soud is very poor how can i improve it?

cgreening, 2020-10-25:
Make sure when you are importing into Audacity you get the settings correct - the Encoding, Byte Order, number of channels and sample rate all need to match what you have recorded.

* Encoding: Signed 16 bit PCM,
* Byte Order: Little Endian
* Channels: 1 Channel (if you are *only* capturing the left or right channel), 2 Channel (if you are capturing *both* left and right)
* Sample rate: 16kHz sample rate

closed 2021-02-09 by cgreening