atomic14/esp32_audio #7: Audio Issue: Audio file received from the ESP32 with INMP441 microphone is distorted

vjsyong, 2020-10-30:
Hi,

I currently am using the INMP441 microphone connected to the ESP32 as per the wiring diagram on the github page. I have been able to get the esp32 to connect over wifi to stream the i2s data back. However, I am running into an issue with the quality of the audio received. I have been importing the audio into audacity with the setting.

1. 16 bit signed PCM
2. 16000hz sampling rate 
3. Mono channel. 

While I am able to hear most of the high level features of the audio, the audio file is generally very distorted and has lots of hissing and popping noises in the background. From watching your previous videos, my first intuition was that it might be a power supply issue, so I added a 1mF capacitor between VCC and GND in order to try to smooth out the electrical noise but to no avail.

Here is sample of the audio received from the ESP32 in a raw format. I have recorded two samples, 1 at 44100hz and the other at 16000hz. The 44100hz sample is a little better but still heavily distorted.

[audiosamples.zip](https://github.com/atomic14/esp32_audio/files/5462496/audiosamples.zip)

StuartIanNaylor, 2020-12-09:
https://drive.google.com/open?id=1FQkvVXz-4xSGMut8JCYTe96_uW_Absuy
Stereo 16bit 44100 raw 2x INMP441using https://github.com/atomic14/ICS-43434-breakout-board

I did get 1x INMP441 that just made this horrid cyclic clicking and replaced it was fine so must be defective.
They are 3.3v devices but sounds like yours is defective

jonaslimads, 2021-01-09:
I was able to remove some noise and hissing by changing to these pins:

```c
i2s_pin_config_t i2sPins = {
    .bck_io_num = 2,
    .ws_io_num = 15,
    .data_out_num = I2S_PIN_NO_CHANGE,
    .data_in_num = 13};
    ```

StuartIanNaylor, 2021-01-11:
I2S is a pure digital signal not analogue you sure you just didn't have your connections right?

closed 2021-02-09 by cgreening

badyast, 2022-02-16:
My raw audio sounds like yours. In my case i solved it using a powerbank as powersupply.