atomic14/esp32_audio #8: Noisy audio

caioyugo123, 2020-11-19:
Hello, I've been trying to capture audio and send it via wifi just as described in the project. But, as soon as I play the RAW file, it is just a noisy audio without a trace of any captured voice. I checked over the code but didn't see anything wrong, although I'm very new to programming in general. I'll be linking the audio acquired via wifi. I'll be very much appreciated for any help, thanks.

[adc.zip](https://github.com/atomic14/esp32_audio/files/5564091/adc.zip)

cgreening, 2020-11-19:
That's very strange. I would expect to see at least some recognisable audio.

Can you show me how the microphone is wired up? Which microphone board are you using?

caioyugo123, 2020-11-19:
I am currently using a esp32 with a max9814 as the image attached below shows. I did a pass low filter, four 330ohm resistors in parallel with four 100microF capacitors also in parallel, in order to supply a more steady voltage in the Vpp pin. But it didn't change much as the audio shown. 

[image]

cgreening, 2020-11-19:
I can't quite follow your low pass filter - I assume you get the noise even when you directly connect Vdd to 3v3?

You can try connecting the gain pin to 3v3 as well which will reduce the amount of gain the microphone is applying.

caioyugo123, 2020-11-19:
I meant by doing as follows:

[image]

I'll try connecting the gain pin.

caioyugo123, 2020-11-19:
Ok, so I tried but it did not work. But, I noticed that during a certain time the audio caption worked. I was measuring voltage and poking the MAX9814 during the recording. Even though, I am not aware the reason why it worked.

[adc (2).zip](https://github.com/atomic14/esp32_audio/files/5568723/adc.2.zip)

cgreening, 2020-11-19:
Can you try running the `loop_sampling` project? This will just output the value read from the ADC on the serial output.

In a quiet room, this should give you around 1.25v and it should range from about 0.25v to 2.25 volts when there is sound.

This will let us quickly check if the microphone is output sensible values.

caioyugo123, 2020-11-19:
Ok, I just uploaded to the ESP32 and then monitored the serial monitor.

[serial monitor.txt](https://github.com/atomic14/esp32_audio/files/5568907/serial.monitor.txt)

cgreening, 2020-11-19:
So, unless you are in a very noisy room this doesn't look good.

We should be seeing around 1250 for the mV values, but instead we are seeing some quite extreme values.

I would double check all your wiring and solder connections. It's also worth trying to power the the microphone from a battery as well just in case it's noise on the power line causing the issues. Two AA batteries will work.

caioyugo123, 2020-11-19:
Unfortunately, it is not the case. Maybe I'll be testing with the loop sampling and will report any progress.
Also, I'll try buying these AA batteries.

StuartIanNaylor, 2020-12-09:
Is it just me or is out connected to gnd? Its hard looking at a picture.

closed 2021-02-09 by cgreening