atomic14/esp32_audio #9: WAV file to SD card

7doctor7, 2020-11-26:
Hi!
First of all - your solution is magic, thanks for your work!))
Can you please provide an example of how I can write ADC samples as to *.wav file to SD card?

BlackLibertyOne, 2020-11-29:
I'll second this request 👍

jake-is-ESD-protected, 2021-02-04:
Currently trying the same thing, your code is a goldmine for well thought-out C++ practices. However, writing to SD comes with a few timing and buffer issues which are quite hard to spot if you don't know how SD-cards work, e.g. it can only write 512 bytes at a time, how is a wav-file structured/padded and giving your SD-card large buffers at slower rates, otherwise the SD-card will slow down the reading of samples. On that note, i don't quite understand your "i2sReaderTask": It features a xQueueReceive from a queue that never stored anything, since i can't find a "xQueueSend". Im guessing you are using it as a method of synchronisation, since just polling for samples can fluctuate wildly in timing and accuracy. Sadly, most tutorials online use that method. But how does it work in your case? The queue should be filled with an event, a notification so to speak, but what triggers this? I'd expect a feature like with the STM32 HAL, where you would configure a register to throw an event into a pre-named queue, but i can't find that in your code. I'd really love to hear you answer and discuss a bit, as well as with other users. Cheers

cgreening, 2021-02-04:
Hi

The queue comes from this line of code in I2SSampler.cpp line 63.

```c++
i2s_driver_install(m_i2sPort, &i2sConfig, 4, &m_i2sQueue);
```

It's not documented very well (if at all) but if you dig into the code here - https://github.com/espressif/esp-idf/blob/master/components/driver/i2s.c

You can see on line 985-991 they create a queue for you.

Events are pushed onto the queue by the i2s driver. The one we are interested in is the event it sends when it has filled one of the DMA buffers - which is `I2S_EVENT_RX_DONE`.

In my code there are two layers of buffering - we have the DMA buffers which the i2s driver reads samples into. Every time one of those gets filled our `i2sReaderTask` wakes up and empties the contents over to our audio buffer.

I've tried to decouple the writing of the samples from the DMA by having my own quite large audio buffers. So we can have a pretty fast function to keep pulling the samples from the DMA buffers. And a much slower function to actually send the samples somewhere.

We leave the reading of the samples to the i2s device driver and DMA so we should get a pretty consistent sampling rate. However, this does mean that there is some latency in the system. This will depend on the size of the DMA buffers you configure the i2s driver with.

* Smaller buffers = lower latency - but at the cost of having to empty the buffers more often so more CPU load
* Larger buffers = higher latency - but lower CPU load

Hope that helps!

cgreening, 2021-02-04:
I need to actually get an SD card breakout board - which just thinking about makes me think that I probably have one somewhere!

I'll dig through the boxes and see.

jake-is-ESD-protected, 2021-02-04:
ooh so that's the "register" - thing i mentioned in reference to the STM32 HAL. What a weird place to put such a configuration. Very newbie-hostile. Goes to show that i don't know the espressif i2s source code too well yet. Thank you for your quick and well explained answer. My code now jumps to a breakpoint set inside the if-statement which asks for the queue-event. Would you recommend pinning the reader task to a core and manage everything else over the other or is freeRTOS more efficient in managing itself freely? 
I didn't know about the open source code up until half an hour ago when a user on stackoverflow pointed it out. I asked similar questions there. I hope i can progress with my code and then open a repo of my own, since working with esp32 and i2s is very cool but still kind of obscured.
Again thanks for the quick advice and your videos, they sure do help de-mystify the whole topic

cgreening, 2021-02-04:
To be honest, it's not really documented anywhere and you shouldn't have to go spelunking through the code to find how it works!

Regarding pinning to a core - it really depends on what else you have going on. Arduino puts a lot of stuff (WiFi and Bluetooth) on core 1, so it's often good to avoid that and pin your own stuff on core 0. But I've also heard people say not to bother and let freeRTOS sort it out - so it may be a case of trying it and see if it makes any difference.

You should definitely put your code up when it's done - it's great to have a portfolio of stuff you've worked on that you can show people.

jake-is-ESD-protected, 2021-02-09:
Hi again, i figured i'd notify you or other users since you helped me so quickly and kindly. I've created a repository for the entirety of my fieldrecorder-project here: https://github.com/jake-is-ESD-protected/I2S-MEMS-mic-field-recorder
Maybe someone is interested in this unfinished code, which i will be updating every few days. The saving of wav-files works, but comes with every audio-issue i could think of...

- very low gain
- negative offset
- loss of samples (short timing intervals)
- loss of samples and replacement with 0s (long timing intervals)
- loss of every second sample it seems, because the recording is twice as fast as the RL-sound but is not pitched higher

It's my job the next few days to figure out these issues and fix them, although the sheer amount of knobs and gears i could turn here are a bit overwhelming. The best combination of buffer-size, speed and program-flow is out there and i'm trying to find it.

cgreening, 2021-06-10:
There's a good example here as well:

https://github.com/atomic14/esp32_sdcard_audio

closed 2021-06-10 by cgreening