atomic14/esp32_audio #11: No Raw files saved?

abdelsha, 2021-01-13:
Where does the raw file save? I looked at the folders and couldn't find any adc.raw

cgreening, 2021-02-08:
Hi Abdelsha,

Did you manage to track down your issues?

They should be written into the `server` folder.

When the server is running it should log out that it is receiving audio data.

Thanks
Chris

abdelsha, 2021-02-08:
Hi Chris, 

No, I wasn't able to track the issue. The esp32 is connected to wifi and everything seems fine from its end. but no data is being saved in the server folder, even after setting up everything on the computer side.

Regards,
Amr

cgreening, 2021-02-09:
I've added some more checks to the code - the ESP32 should now ping the server every 5 seconds so you can tell if it can see the server.

abdelsha, 2021-02-15:
I am using the esp32_audio/i2s_sampling/ folder path and then updating the wifi credentials. The folder where the check was done was a different folder. Am I using the wrong code?

abdelsha, 2021-02-19:
do you know how to write a .wav audio file before sending it?

peymanrmi, 2021-02-27:
i have same problem
no raw file saved in server folder
i start the server at localhost:5003

I've added some more checks to the code - the ESP32 should now ping the server every 5 seconds so you can tell if it can see the server.

where is the code that ping the server??!!!