atomic14/esp32_audio #12: no audio / scope data OK

optimaxch, 2021-02-08:
hi,

[env:pico32]
platform = espressif32
board = pico32
framework = arduino
upload_port = COM5
monitor_speed = 115200

    .bck_io_num   = GPIO_NUM_26,             // bit clock
    .ws_io_num    = GPIO_NUM_32,             // word select L/R
    .data_out_num = GPIO_NUM_25,           // data

upload OK, no error messages.

adafruit MX98357
[image]

  danke aus österreich
  rudi

cgreening, 2021-02-08:
Hi Rudi,

Did you upload the filesystem image - that contains the WAV file - I'm not sure if there's much error checking on opening the file.

Thanks
Chris

optimaxch, 2021-02-08:
hi chris,
yes i tried with two boards, tried sine wave and sampleWAV  and uploaded image with tool
(this is adafruit breakout )
i will try sparkfun next....
[image]

thanks from austria
  rudi

cgreening, 2021-02-17:
Sorry Rudi - that was a silly question on my part - of course you've uploaded the file system - otherwise you wouldn't see any waveforms on the scope.

You have me stumped - can you post a photo of the connections going into the MAX? It might be worth trying the tie the gain pin and the SD pin to high or low - just make sure there's nothing weird happening from having them floating.

And you've definitely go power going the MAX breakout board?

optimaxch, 2021-02-19:
hi chris,
took me long to adapt your work to ESP32-pico, using platformio.
actually i needed to open a new project, and copied your files and libs to  project directory.
using your example i2s_output.
i am running a sparkfun MAX98357  on 5V , gain pin grounded, with crisp sound - very clear!
when adding scope to speaker output  - sound is gone...

i am looking for changing gain per software command ?
where to change the    gain setting in  lib ?

very satisfied , THANKS 
good day & health
greetings to london from villach
 rudi

optimaxch, 2021-02-20:
hi chris,
took me too long to figure out the correct audio settings, when adding own  .wav file:

recording
audacity
audio  host  :  Windows WASAPI 
track        :  mono
preferences  quality:   8000Hz / 16 Bit

file export  wav
save as   :  wav (Microsoft)
encoding  :  Signed 16-bit PCM

maybe this helps, other audiophiles...
( its interesting you original sample file is  8000Hz / 32bit float )

can you help to play different  .wav files one after next ?
i tried in the loop  - without successssssss

  thanks
  rudi

optimaxch, 2021-04-30:
hi chris

i am unable to play separate    .wav files     one after next..
please can you help.
i like to create an  short announcement voice, speaking a text according an event.
for example  "power 1 "   power 2"...
- and i am unable to place your code inside   loop  .
[audio.txt](https://github.com/atomic14/esp32_audio/files/6403555/audio.txt)

 thanks from austria
  rudi

cgreening, 2021-04-30:
Hey Rudi,

It might be worth having a look at the code I wrote for recording and playing from SD Card which I've updated to work with SPIFFS.

https://github.com/atomic14/esp32_sdcard_audio/tree/main/arduino-wav-sdcard

Jusy uncomment the line in config.h to switch on SPIFFs instead of SD Card.

It's quite a bit simpler and would work well in loops.

I need to come back and simplify the example code here.