atomic14/esp32_audio #13: WifiClient not connect

Lubinari, 2021-02-16:
Hello, I found your project interesting, because I'll use a MAX9814 in the future, and I'm researching more about I2S too. However, I'm having problems connecting to my WIFI network. It's causing this error: 
[E][WiFiClient.cpp:258] connect(): socket error on fd 54, errno: 113, "Software caused connection abort"
So far I don't know how I can solve it, if you can help I'll be grateful.

cgreening, 2021-02-17:
Hi there,

Couple of things to check

- is the ESP32 connecting to the WiFi network? You should see a message: "Starting up"
- Have you updated the IP-Address of the server it is trying to connect to? There are two lines of code that need to be modified in `main.cpp` and updated to the IP address of the machine you are running the server on.
```
// replace this with your machines IP Address
#define ADC_SERVER_URL "http://10.0.1.18:5003/adc_samples"
#define I2S_SERVER_URL "http://10.0.1.18:5003/i2s_samples" 
```
- And final check - are you running the server?

Thanks
Chris

Lubinari, 2021-02-17:
All these steps have already been done, but the same error persists. 
Tanks, David

johnkrmr, 2021-04-18:
Hi David,

I was stuck on the same problem. It's probably a firewall issue. I guess you could try messing with your firewall's settings or disable it completely.
A (probably more safe) workaround that worked for me was to create an access point (AP) on my laptop with the firewall zone set to 'trusted' (on GNU/Linux) and then connected the ESP32 to the AP.
Make sure set the correct the ip address since it'll be different in the AP network.

Best regards,
John

cgreening, 2021-08-23:
Thanks @johnkrmr that's very useful advice. I'm going to close this issue since it's gone quiet.

closed 2021-08-23 by cgreening