atomic14/esp32_audio #16: Reading second ADC channel

SpirosMakris, 2021-04-26:
Hello,

first of all I would like to thank you for this code! I'm using it to sample from an analog microphone (ADC sampler code path).

It works, although I'm not quite sure what the output data format is for the samples (MSB vs LSB), but the main proble  I'm facing is that in another task I'm using Emon library to sample another sensor using ADC0 but on another channel.

Turns out this makes the I2S/ADCSampler give back a single value of 0. I tried using `i2s_adc_enable` and `i2s_adc_disable` around the `i2s_read` but it has no effect.

Even if I don't use the Emon library, and simply use `adc1_get_raw()` the same thing happens so that makes me think it's some sort of ADC config issue.

Do you have any thoughts or ideas why this may happen?

Thank you in advance,
Spiros

SpirosMakris, 2021-04-26:
Correction, `adc1_get_raw()` looks ok, so it's probably something Emon lib does.
From a cursory glance at the code it uses `analogRead()` so maybe  something is going wrong there.

cgreening, 2021-04-26:
I think the code in analogRead does quite a lot of monkeying around. So it's possible that it's interfering with other ADC functions.

I don't think there should be any interference between reading with I2S and getting values manually, but there could be some weird interactions going on.

The esp32 forums are really good for this kind of weird thing - lots of smart people there - https://www.esp32.com/viewforum.php?f=23

SpirosMakris, 2021-05-06:
Hi   @cgreening,
sorry for the delayed reply on my part. Those were my thoughts also.
In the end I went with an I2C mic (the IMLP one) directly and it works like a charm.

So I am closing this issue.
Thanks everyone.

closed 2021-05-06 by SpirosMakris