atomic14/esp32_audio #17: Play audio only once (dac i2s)

AljoshaVieth, 2021-06-09:
How could I play an audio file only once and not in a loop? 
I tried to modify the code in DAC_i2S_OUTPUT to play a .wav file only once instead of in a loop.
My solution is rather radical, I´m simply killing the writer task and uninstall the i2s driver after a period of time (duration of the audio).

```c++ 
    delay(audio_length); //wait till audio is fully played
    if (writerTaskHandle != NULL) {
        vTaskDelete(writerTaskHandle);
        Serial.println("AudioTask killed!");
        i2s_driver_uninstall(I2S_NUM_0);  // stop & destroy i2s driver (audio ends)
    }
```

The solution works, but sometimes I get problems with the task, even though I wait long enough before killing it:
```
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:4101 (xTaskPriorityDisinherit)- assert failed!
abort() was called at PC 0x400880a7 on core 1```

My final goal is to have several buttons which each play another .wav (not simultaneously).

cgreening, 2021-06-09:
I need to rework these examples to make them a bit simpler.

You might find this example a bit easier to work with: https://github.com/atomic14/esp32_sdcard_audio

AljoshaVieth, 2021-06-10:
Thank you very much. I tried https://github.com/atomic14/esp32_sdcard_audio but I´m not sure how to output the audio via dac instead of i2s

cgreening, 2021-06-10:
It should just be a case of editing the config.h file and commenting out `#define USE_I2S_MIC_INPUT`

cgreening, 2021-06-10:
Sorry - not a very wll named #define - now that I look at it! I'll change it.

cgreening, 2021-06-10:
Let me add in the code to do it - I've just realised it's not wired up.

cgreening, 2021-06-10:
I've updated the code - I won't be able to test until tonight.

AljoshaVieth, 2021-06-10:
Thank you very much, I will test it. 
It is a project of my university. We're not all very experienced in this field, but we're trying our best. There should be several buttons that play bird sounds, you should be able to press a button and the corresponding wav file should be played. Unfortunately, we don't have much hardware and are using the [micro-bit speaker](https://www.waveshare.com/wiki/Speaker_for_micro:bit) on an esp32  and are therefore dependent on the dac.

So you are really saving our days, thank you very much!

cgreening, 2021-06-10:
Let me know how you get on. You should be able to reuse the code from that project.

I will be updated these examples to make them easier to understand/use, but that will take some time.

AljoshaVieth, 2021-06-17:
Again, thank you very much, I managed to archive exactly what I was looking for with your code and minor adjustments.

closed 2021-07-08 by AljoshaVieth

cgreening, 2021-07-08:
That's great! Congratulations.