atomic14/esp32_audio #24: Spiky audio capture

Ma100Dev, 2021-12-30:
Hello!
I am trying to listen to the audio recorded by the ESP32, but it just sounds like a bunch of clicks. I looked at the raw audio data with Audacity and the audio waveform indeed is almost exclusively formed out of spikes. What could be causing this and how would one go about fixing it? I experimented with whether lowering the gain on the MAX9814 module's gain by tying its gain pin high (40dB gain), but it did not help.
[image] |
|:--:|
| The raw audio data in Audacity |

frmunozForcast, 2022-06-20:
@Ma100Dev I know this is an old issue, but I'm facing the exact same problem, did you managed to solve it?

Ma100Dev, 2022-06-21:
No. There was never an answer to this issue and I didn't manage to figure it out myself.